graycatlabs/PyBBIO #58: TypeError: must be string, not int

alyyousuf7, 2015-03-14:
In a freshly installed Debian (2015-03-01) on Beaglebone Black Rev A6A, the encoder_test.py example shows this error:

```
Traceback (most recent call last):
  File "encoder_test.py", line 24, in <module>
    run(setup, loop)
  File "/usr/local/lib/python2.7/dist-packages/PyBBIO-0.9.3-py2.7-linux-armv7l.egg/bbio/bbio.py", line 60, in run
    loop()
  File "encoder_test.py", line 22, in loop
    delay(1000)
  File "/usr/local/lib/python2.7/dist-packages/PyBBIO-0.9.3-py2.7-linux-armv7l.egg/bbio/common.py", line 36, in delay
    time.sleep(ms/1000.0)
TypeError: must be string, not int
Error in atexit._run_exitfuncs:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/PyBBIO-0.9.3-py2.7-linux-armv7l.egg/bbio/bbio.py", line 21, in bbio_cleanup
    for cleanup in ADDITIONAL_CLEANUP:
TypeError: must be string, not int
Error in sys.exitfunc:
Traceback (most recent call last):
  File "/usr/lib/python2.7/atexit.py", line 24, in _run_exitfuncs
    func(*targs, **kargs)
  File "/usr/local/lib/python2.7/dist-packages/PyBBIO-0.9.3-py2.7-linux-armv7l.egg/bbio/bbio.py", line 21, in bbio_cleanup
    for cleanup in ADDITIONAL_CLEANUP:
TypeError: must be string, not int
```

sahmad98, 2015-03-14:
Code is wrong at time.sleep(ms/1000), is ms previously defined. Check for
that because sleep function takes int.

alyyousuf7, 2015-03-14:
I have not changed anything in the example. It is in integer already. The error says "must be String" -- I'm also confused at this.
Here is the code
```
#!/usr/bin/env python
"""
 encoder_test.py
 Rekha Seethamraju

 An example to demonstrate the use of the eQEP library
 for PyBBIO.

 This example program is in the public domain.
"""
from bbio import *
from bbio.libraries.RotaryEncoder import RotaryEncoder

encoder = RotaryEncoder(RotaryEncoder.EQEP2b)

def setup():
  encoder.setAbsolute()
  encoder.zero()

def loop():
  print "encoder position : \t"+encoder.getPosition()
  delay(1000)

run(setup, loop)
```

alexanderhiam, 2015-03-15:
Hmm... this is an odd one. Seems that either the RotaryEncoder library itself or the sysfs C extension it uses to read/write the eQEP driver files is the source of the problem (time.sleep most certainly doesn't take a string!). I'm looking into it...

KirkAtWork, 2015-04-07:
Just letting you know I'm having the exact same problem with the same code
I'm on Beaglebone Black Rev C.  Version is:
Linux beaglebone 3.8.13-bone70 #1 SMP Fri Jan 23 02:15:42 UTC 2015 armv7l GNU/Linux

Any new ideas on this yet?  I'm stuck

alexanderhiam, 2015-04-08:
OK got it, I was missing a type check in a C extension, and the TypeError was popping up at a weird time.

closed 2015-04-08 by alexanderhiam