graycatlabs/PyBBIO #59: PEP 8 complience

asomov, 2015-03-31:
It is convinient for an Open Surce project to stick to PEP 8.
https://www.python.org/dev/peps/pep-0008/

If we agree to follow PEP 8 (4 spaces, for instance) I can apply the changes myself.

alexanderhiam, 2015-04-07:
Meh, I'm not a fan of the 4-space indent, it takes up way to much horizontal space IMO. Besides that it pretty much lines up with the coding style I've used here.

asomov, 2015-04-08:
Well, you can use any style you want, but please take it into account that if you expect others to contribute they will be always confronted with this deviation from the standard (or any other deviation). 
If the decision to deviate is explicit, it is better to expose is publicly on the wiki page for contributing.

alexanderhiam, 2015-04-08:
PEP 8 is by no means 'the standard', it's just what the Python team chose for their code. 

It probably is a good idea to make it clear in the wiki though.

closed 2015-04-09 by alexanderhiam