graycatlabs/PyBBIO #63: 3.14 support

alexanderhiam, 2015-03-31:
Since the BeagleBone's 3.14 kernel doesn't include capemgr, PyBBIO will need a new method of dynamic pinmuxing. This could be done through the existing cape-universal (https://github.com/cdsteinkuehler/beaglebone-universal-io), which would also require the user to enable overlays for any peripheral subsystems they want to use.

anujdeshpande, 2015-04-02:
Check the comments on this particular post : https://plus.google.com/+EliasBakken/posts/Uhejy5jLuv4
panto says patches are coming through. Could stick to capemgr

alexanderhiam, 2015-04-06:
That's 4.0 though. AFAIK BeagleBone's will still be shipping with 3.14 for now.

alexanderhiam, 2015-07-07:
Closing this for now as it looks like the next BeagleBone release will have 4.1 instead of 3.14.

closed 2015-07-07 by alexanderhiam