graycatlabs/PyBBIO #65: Update pinout diagram

alexanderhiam, 2015-04-02:
The pinout diagram (https://github.com/graycatlabs/PyBBIO/wiki#beaglebone-pinout) needs to be updated to include all the currently supported subsystems. 

This should be done before the v0.9.4 release so it can be included in a pdf manual.

anujdeshpande, 2015-04-02:
So SPI, I2C remain. Do you think there is a need to show that all the pins can be interrupts too ? Also, what s/w did you use to create the image ? Looks pretty neat. I don't mind finishing up this. Doesn't look like much effort.

alexanderhiam, 2015-04-05:
eQEP as well.

It might be nice to add an index, in which case it could indicate the GPIO pins as something like 'GPIO/Interrrupt'.

I used Inkscape, but it's just an SVG. My site doesn't want to serve up SVGs right now for some reason, but you should be able to download it from here: https://gist.github.com/alexanderhiam/24ce75458b4b13c4bf9f

alexanderhiam, 2015-04-20:
I beat you to it!

closed 2015-04-20 by alexanderhiam

anujdeshpande, 2015-04-20:
@alexanderhiam is faster :D 
[image]