graycatlabs/PyBBIO #66: PWM out and RES_16BIT problem

lucalenardi, 2015-04-11:
Hello and thanks for the hard work.

I'm testing the PWM output (PWM1A) to regulate an LED brightness:

* Issuing a `pwmWrite(PWM1A, 255)` works as expected and the LED lights up to full brightness (allowed by the voltage divider);
* Issuing a `pwmWrite(PWM1A, 65536, RES_16BIT)` doesn't work, it doesn't log any error and the LED stays turned off. Changing values doesn't make any difference.

Tested with both PWM1* and PWM2*. 
Any clues?

nicktime, 2015-08-31:
Hi, e.g.
```
duty_ns = int(value * (period_ns/resolution))
```
https://github.com/graycatlabs/PyBBIO/blob/master/bbio/platform/beaglebone/pwm.py#L37
looks like is using integer division, in python 2.7 "/" for ints returns int.
Is it intentional? 
```
from __future__ import division
```
would help.

alexanderhiam, 2015-09-02:
Interesting, looking through that file (it's been a while since I wrote that) it looks like there are a few divisions that I should reassess. Just importing the new division operator could potentially cause trouble elsewhere since I tend to use the default floor division intentionally. 

This module is next on my list to move to a C extension to speed things up, so I'll probably leave it as is for now and get to doing that ASAP.