graycatlabs/PyBBIO #77: analogWrite() not always sets the value

nicktime, 2015-07-17:
Hi,

looking at the code I noticed pwm's analogWrite() just reuses previous period:
```
    period_ns = int(sysfs.kernelFileIO('%s/%s' % (helper_path, PWM_PERIOD)))
```
which means if e.g. pwmWrite() was used to set some low frequency, next analogWrite() won't set output correctly.

alexanderhiam, 2015-07-17:
That's reading the period of the PWM waveform, i.e. the total time of one full cycle. The duty cycle is set by writing the duration of the high (or low if the polarity is inverted) pulse, so that has to be calculated based on the PWM period. In other words that should be correct as is.

nicktime, 2015-07-19:
Oh, sorry, missed that the cycle can be asymmetric. You're right.

closed 2015-07-19 by nicktime

nicktime, 2015-07-19:
On the other hand e.g.
```
duty_ns = int(value * (period_ns/resolution))
```
https://github.com/graycatlabs/PyBBIO/blob/master/bbio/platform/beaglebone/pwm.py#L37
looks like is using integer division, in python 2.7 "/" returns int.
Is it intentional? 
```
from __future__ import division
```
would help.