graycatlabs/PyBBIO #79: BMP183_test.py does not gie correct pressure output

EngrPradeep, 2015-07-31:
I am not able to get the correct output from the BMP183 pressure sensor. However the temperature readings are correct. But if I use the Adafruits library using on arduino for the same connections, same sensor, I get the correct readings. I tried to modify the code in the similar fashion as Adafruit library but it still gives me the same (wrong pressure reading) output. I am using BBB with python 2.7 on debian. 

Any suggestion will be very much helpful.

Thank you in advance!

EngrPradeep, 2015-08-03:
Apparently looks like there is a bug in the BMP183 library. The program runs and gives the output even after BMP183.py file removed from the BMP183 library folder. 

I guess there is some error in the following lines of BMP183_test.py file:

from bbio import *
# Import the BMP183 class from the BMP183 library:
from bbio.libraries.BMP183 import BMP183  

Still trying to figure out how the code is written. Any help will be greatly appreciated. (I am novice to python and BeagleBone)

alexanderhiam, 2015-08-05:
Sorry, haven't had a chance to look into this yet, but will check it out ASAP. Not sure what's going on, I'm pretty certain I was getting correct values when I tested it, or else I wouldn't have pushed it.

When you install PyBBIO it installs the package to `/usr/local/lib/python2.7/dist-packages/PyBBIO-*.egg`, so you'd have to delete the file from there. If you did that then it would throw an import error.

Mausy5043, 2016-01-17:
I have the same problem. I get perfectly normal temperatures. But the pressure is at 25700 Pa and I'm pretty sure I'm not almost out in space :wink:

I'm on a BBB using `/usr/local/lib/python2.7/dist-packages/PyBBIO-0.10-py2.7.egg-info`

Mausy5043, 2016-01-17:
I've done a very quick regression of my data and the numbers from @EngrPradeep and found this relation:
```
P = -0.1 * p +105000  (roughly)
```
Where `P` is the expected pressure and `p` the reported pressure.
The gain and offset look interestingly familiar. Hope this is helpful.

Mausy5043, 2016-01-17:
I think I found the problem.

The calculation of [B3](https://github.com/graycatlabs/PyBBIO/blob/master/bbio/libraries/BMP183/BMP183.py#L108) is incorrect.

If you follow the calculation using the data from the datasheet your code results in a value for B3 which is off by a factor of approx 4.3.

The formula on the datasheet reads exactly the same as in your code: `b3 = ((self.cal_AC1*4+x3) << oversampling + 2) / 4`. The interpretation by Python is to shift left by `oversampling+2`

If I change the code to:
`b3 = (((self.cal_AC1*4+x3) << oversampling) + 2) / 4`
the result is correct.

Mausy5043, 2016-01-17:
Proposed a PR (#96) to fix it.

alexanderhiam, 2016-01-20:
Merged - thanks @Mausy5043!

closed 2016-01-20 by alexanderhiam

Mausy5043, 2016-01-21:
No problem.
:+1: