graycatlabs/PyBBIO #80: Support for Beaglebone devices running 4.1 kernel

shaunmulligan, 2015-08-05:
Since the new capemgr has now been included in the latest 4.1 kernel, I was wondering if there was a plan to support it?

linux repo: https://github.com/beagleboard/linux/tree/4.1
overlay repo: https://github.com/RobertCNelson/bb.org-overlays

As far as I can see the major changes are in the file path
slots:
/sys/devices/platform/bone_capemgr/slots

I assume the BB foundation will be moving towards this kernel in the near future, so just interested in finding out if this lib will support it?

alexanderhiam, 2015-08-05:
That's one of the next things I was going to work on. It's now official that the 4.1 Kernel that will ship will have the universal-io overlay loaded by default, so the plan has two parts:

1. add support for universal-io to use instead of the current custom overlay approach
2. detect when universal-io isn't loaded (e.g. another cape overlay is loaded), then fallback to the custom overlay approach, modified for 4.1

Besides the path, there's also some minor changes to the device tree, so the custom overlays will need some changes. 

Ideally PyBBIO will still support the 3.8 kernel for a while as well.

shaunmulligan, 2015-09-11:
Hi there, any update on the 4.1 kernel support?

alexanderhiam, 2015-10-06:
Moved this into to separate issues (issue #86 and #87), I'll be working on them shortly...

closed 2015-10-06 by alexanderhiam