graycatlabs/PyBBIO #85: Memory leak in SPI transfer

KirkAtWork, 2015-09-29:
In the past there was a memory leak in the SPI0.read function.  It was fixed.

Now I'm seeing a problem with the SPI0.transfer function.
The Python code below demonstrates the problem.
The amount of memory "leaked" is proportional to the number of words transferred.
Un-comment one of the lines in the for loop to observe the problem.
Sample output is included below also.

Thanks very much for taking a look,
Kirk Mitchell
Kollmorgen

******************** python script to demonstrate the memory leak **************

#pybbioMemoryLeakV102.py
#PyBBIO SPI memory leak demonstration

from bbio import *
import resource

# initialize the SPI to talk to the ADXL312 accelerometer chip
def setup():
    SPI0.begin()
    SPI0.setClockMode(0,3)
    SPI0.setMaxFrequency(0, 5000000)
    SPI0.setBitsPerWord(0, 16)

# read blocks of words from the accelerometer
def loop():
    for j in range(10):
        print "*******Reading 100000 samples, block=",j
        mem_used = resource.getrusage(resource.RUSAGE_SELF).ru_maxrss
        print 'Memory usage( kB)=', mem_used

        for i in range(100000):
            #The following 3 lines of code demonstrate the memory leak (use only 1 at a time)
            
            #spiReadData = SPI0.read(0, 1)  #this line does not leak
            #spiReadData = SPI0.transfer(0, [0, 0, 0, 0]) #this line leaks
            spiReadData = SPI0.transfer(0, [0, 0, 0, 0, 0, 0, 0, 0])    #this line leaks twice as fast as the line above
            
        delay( 1000 )

    print "Cleaning up & exiting"
    SPI0.end
    stop()
    
run(setup, loop)

*********** Output from 2 runs of the above code follows ****************************

root@beaglebone:/kirkpy# python pybbioMemoryLeakV102.py
*******Reading 100000 samples, block= 0
Memory usage( kB)= 4372
*******Reading 100000 samples, block= 1
Memory usage( kB)= 10552
*******Reading 100000 samples, block= 2
Memory usage( kB)= 15280
*******Reading 100000 samples, block= 3
Memory usage( kB)= 20008
*******Reading 100000 samples, block= 4
Memory usage( kB)= 24732
*******Reading 100000 samples, block= 5
Memory usage( kB)= 29456
*******Reading 100000 samples, block= 6
Memory usage( kB)= 34184
*******Reading 100000 samples, block= 7
Memory usage( kB)= 38908
*******Reading 100000 samples, block= 8
Memory usage( kB)= 43636
*******Reading 100000 samples, block= 9
Memory usage( kB)= 48360
Cleaning up & exiting

root@beaglebone:/kirkpy# python pybbioMemoryLeakV102.py
*******Reading 100000 samples, block= 0
Memory usage( kB)= 4372
*******Reading 100000 samples, block= 1
Memory usage( kB)= 15276
*******Reading 100000 samples, block= 2
Memory usage( kB)= 24732
*******Reading 100000 samples, block= 3
Memory usage( kB)= 34184
*******Reading 100000 samples, block= 4
Memory usage( kB)= 43636
*******Reading 100000 samples, block= 5
Memory usage( kB)= 53084
*******Reading 100000 samples, block= 6
Memory usage( kB)= 62536
*******Reading 100000 samples, block= 7
Memory usage( kB)= 71988
*******Reading 100000 samples, block= 8
Memory usage( kB)= 81440
*******Reading 100000 samples, block= 9
Memory usage( kB)= 90892
Cleaning up & exiting

alexanderhiam, 2015-09-30:
Thanks for another good catch, Kirk! It's fixed now on Github and I'll push another release to PyPI soon.

closed 2015-09-30 by alexanderhiam

KirkAtWork, 2015-09-30:
Sorry about the weird bold text in the post.  Not sure what happened.
I really appreciate your quick response to this.  I'll test it and let you know how it goes.
Thanks again,
Kirk