graycatlabs/PyBBIO #90: example blink.py error with beaglebone black

Ldsrc2008, 2015-11-04:
Hi, Alex,

I have installed PyBBIO 0.10 on a beaglebone black with angstrom 2012.12 kernel 3.8.13, 
after installation I ran python blink.py it give me message below. what did I do wrong? 
root@beaglebone:~/PyBBIO/examples# python blink.py
Traceback (most recent call last):
  File "blink.py", line 7, in <module>
    from bbio import *
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/__init__.py", line 5, in <module>
    from .bbio import run, stop
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/bbio.py", line 10, in <module>
    from .platform import platform_init, platform_cleanup
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/__init__.py", line 6, in <module>
    from beaglebone import *
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/beaglebone/__init__.py", line 9, in <module>
    from i2c import I2C1, I2C2, Wire1, Wire2
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/beaglebone/i2c.py", line 63, in <module>
    Wire1 = I2C1 = I2CBus(1)
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/beaglebone/i2c.py", line 15, in __init__
    super(I2CBus, self).__init__(bus)
SystemError: NULL result without error in PyObject_Call

here are the installation log:
root@beaglebone:~/PyBBIO# ls
CHANGES.md   PyBBIO.egg-info  build               doinstall.sh  setup.py
LICENSE.txt  README.md        code_of_conduct.md  examples      tests
MANIFEST.in  bbio             dist                setup.cfg     tools
root@beaglebone:~/PyBBIO# python setup.py install
Installing PyBBIO...
running install
running bdist_egg
running egg_info
writing requirements to PyBBIO.egg-info/requires.txt
writing PyBBIO.egg-info/PKG-INFO
writing top-level names to PyBBIO.egg-info/top_level.txt
writing dependency_links to PyBBIO.egg-info/dependency_links.txt
reading manifest file 'PyBBIO.egg-info/SOURCES.txt'
reading manifest template 'MANIFEST.in'
writing manifest file 'PyBBIO.egg-info/SOURCES.txt'
installing library code to build/bdist.linux-armv7l/egg
running install_lib
running build_py
running build_ext
creating build/bdist.linux-armv7l/egg
creating build/bdist.linux-armv7l/egg/bbio
creating build/bdist.linux-armv7l/egg/bbio/libraries
creating build/bdist.linux-armv7l/egg/bbio/libraries/RotaryEncoder
copying build/lib.linux-armv7l-2.7/bbio/libraries/RotaryEncoder/rotary_encoder.py -> build/bdist.linux-armv7l/egg/bbio/libraries/RotaryEncoder
copying build/lib.linux-armv7l-2.7/bbio/libraries/RotaryEncoder/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/RotaryEncoder
creating build/bdist.linux-armv7l/egg/bbio/libraries/EventIO
copying build/lib.linux-armv7l-2.7/bbio/libraries/EventIO/eventio.py -> build/bdist.linux-armv7l/egg/bbio/libraries/EventIO
copying build/lib.linux-armv7l-2.7/bbio/libraries/EventIO/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/EventIO
creating build/bdist.linux-armv7l/egg/bbio/libraries/LiquidCrystal
copying build/lib.linux-armv7l-2.7/bbio/libraries/LiquidCrystal/LiquidCrystal.py -> build/bdist.linux-armv7l/egg/bbio/libraries/LiquidCrystal
copying build/lib.linux-armv7l-2.7/bbio/libraries/LiquidCrystal/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/LiquidCrystal
creating build/bdist.linux-armv7l/egg/bbio/libraries/SafeProcess
copying build/lib.linux-armv7l-2.7/bbio/libraries/SafeProcess/safe_process.py -> build/bdist.linux-armv7l/egg/bbio/libraries/SafeProcess
copying build/lib.linux-armv7l-2.7/bbio/libraries/SafeProcess/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/SafeProcess
creating build/bdist.linux-armv7l/egg/bbio/libraries/ADT7310
copying build/lib.linux-armv7l-2.7/bbio/libraries/ADT7310/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/ADT7310
copying build/lib.linux-armv7l-2.7/bbio/libraries/ADT7310/adt7310.py -> build/bdist.linux-armv7l/egg/bbio/libraries/ADT7310
creating build/bdist.linux-armv7l/egg/bbio/libraries/HTU21D
copying build/lib.linux-armv7l-2.7/bbio/libraries/HTU21D/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/HTU21D
copying build/lib.linux-armv7l-2.7/bbio/libraries/HTU21D/HTU21D.py -> build/bdist.linux-armv7l/egg/bbio/libraries/HTU21D
creating build/bdist.linux-armv7l/egg/bbio/libraries/BBIOServer
copying build/lib.linux-armv7l-2.7/bbio/libraries/BBIOServer/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/BBIOServer
copying build/lib.linux-armv7l-2.7/bbio/libraries/BBIOServer/bbio_server.py -> build/bdist.linux-armv7l/egg/bbio/libraries/BBIOServer
creating build/bdist.linux-armv7l/egg/bbio/libraries/GridEYE
copying build/lib.linux-armv7l-2.7/bbio/libraries/GridEYE/GridEYE.py -> build/bdist.linux-armv7l/egg/bbio/libraries/GridEYE
copying build/lib.linux-armv7l-2.7/bbio/libraries/GridEYE/__init__.py -> build/bdist.linux-armv7l/egg/bbio/libraries/GridEYE
[... 72 lines cut ...]
removing 'build/bdist.linux-armv7l/egg' (and everything under it)
Processing PyBBIO-0.10-py2.7-linux-armv7l.egg
removing '/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg' (and everything under it)
creating /usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg
Extracting PyBBIO-0.10-py2.7-linux-armv7l.egg to /usr/lib/python2.7/site-packages
PyBBIO 0.10 is already the active version in easy-install.pth

Installed /usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg
Processing dependencies for PyBBIO==0.10
Searching for serbus==1.0.4
Best match: serbus 1.0.4
Adding serbus 1.0.4 to easy-install.pth file

Using /usr/lib/python2.7/site-packages
Searching for requests==2.8.1
Best match: requests 2.8.1
Adding requests 2.8.1 to easy-install.pth file

Using /usr/lib/python2.7/site-packages
Searching for pyserial==2.7
Best match: pyserial 2.7
Adding pyserial 2.7 to easy-install.pth file

Using /usr/lib/python2.7/site-packages
Finished processing dependencies for PyBBIO==0.10

sjMoquin, 2015-12-02:
Repeated results on my machine:

root@beaglebone:~/PyBBIO# cd /usr/local/lib/PyBBIO/examples/
root@beaglebone:/usr/local/lib/PyBBIO/examples# python blink.py 
Traceback (most recent call last):
  File "blink.py", line 7, in <module>
    from bbio import *
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/__init__.py", line 5, in <module>
    from .bbio import run, stop
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/bbio.py", line 10, in <module>
    from .platform import platform_init, platform_cleanup
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/__init__.py", line 6, in <module>
    from beaglebone import *
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/beaglebone/__init__.py", line 9, in <module>
    from i2c import I2C1, I2C2, Wire1, Wire2
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/beaglebone/i2c.py", line 63, in <module>
    Wire1 = I2C1 = I2CBus(1)
  File "/usr/lib/python2.7/site-packages/PyBBIO-0.10-py2.7-linux-armv7l.egg/bbio/platform/beaglebone/i2c.py", line 15, in __init__
    super(I2CBus, self).__init__(bus)
SystemError: NULL result without error in PyObject_Call

less etc/dogtag:
Cloud9 GNOME Image 2013.09.04

alexanderhiam, 2015-12-03:
@sjMoquin That's also an Angstrom image? I'm not technically supporting Angstrom these days since the Debian builds have been the only images maintained and supported by BeagleBoard.org - you shouldn't have any trouble if you update...

That said, it looks like I'm not catching an error in the I2C C extension, which isn't very good... I'll look into it...