graycatlabs/PyBBIO #91: BBIOServer_test.py  error

Rocketpol, 2015-12-09:
i want test the  BBIOServer_test.py  in my beaglebone Black (Linux beaglebone 3.8.13-bone79 #1 SMP Tue Oct 13 20:44:55 UTC 2015 armv7l GNU/Linux) but when i press Run on my cloud9 the compiler return 

Traceback (most recent call last):
  File "/var/lib/cloud9/examples/PyBBIO/BBIOServer_test.py", line 15, in <module>
    from bbio.libraries.BBIOServer import *
  File "/usr/local/lib/python2.7/dist-packages/bbio/libraries/BBIOServer/__init__.py", line 3, in <module>
    from bbio_server import *
  File "/usr/local/lib/python2.7/dist-packages/bbio/libraries/BBIOServer/bbio_server.py", line 26, in <module>
    os.chdir(ROOT_DIR)
OSError: [Errno 2] No such file or directory: '/root/.BBIOServer'

Someone can help me to solve this problem, i'm new on Beaglebone and also linux machine.

Thanks

alexanderhiam, 2015-12-09:
Hmm, it should have created and populated that directory when PyBBIO installed - try updating:

    # pip install -U PyBBIO

and run it again. If that still doesn't work then see if uninstalling and reinstalling does the trick:

    # pip uninstall PyBBIO
    # pip install PyBBIO

Rocketpol, 2015-12-10:
Thanks for your help i try what you tell me but nothing change this is the result of your command:

root@beaglebone:~# pip install -U PyBBIO
Requirement already up-to-date: PyBBIO in /usr/local/lib/python2.7/dist-packages
Requirement already up-to-date: pyserial in /usr/local/lib/python2.7/dist-packag                                      es (from PyBBIO)
Requirement already up-to-date: requests in /usr/local/lib/python2.7/dist-packag                                      es (from PyBBIO)
Requirement already up-to-date: serbus in /usr/local/lib/python2.7/dist-packages                                       (from PyBBIO)
Cleaning up...
root@beaglebone:~# pip uninstall PyBBIO
Uninstalling PyBBIO:
Proceed (y/n)? y
  Successfully uninstalled PyBBIO
root@beaglebone:~#
root@beaglebone:~# pip install PyBBIO
Requirement already satisfied (use --upgrade to upgrade): PyBBIO in /usr/local/lib/python2.7/dist-pack                ages
Requirement already satisfied (use --upgrade to upgrade): pyserial in /usr/local/lib/python2.7/dist-pa                ckages (from PyBBIO)
Requirement already satisfied (use --upgrade to upgrade): requests in /usr/local/lib/python2.7/dist-pa                ckages (from PyBBIO)
Requirement already satisfied (use --upgrade to upgrade): serbus in /usr/local/lib/python2.7/dist-pack                ages (from PyBBIO)
Cleaning up...
root@beaglebone:~#

And from the compiler i have the same result: 
raceback (most recent call last):
  File "/var/lib/cloud9/examples/PyBBIO/BBIOServer_test.py", line 15, in <module>
    from bbio.libraries.BBIOServer import *
  File "/usr/local/lib/python2.7/dist-packages/bbio/libraries/BBIOServer/__init__.py", line 3, in <module>
    from bbio_server import *
  File "/usr/local/lib/python2.7/dist-packages/bbio/libraries/BBIOServer/bbio_server.py", line 26, in <module>
    os.chdir(ROOT_DIR)
OSError: [Errno 2] No such file or directory: '/root/.BBIOServer'

Please if you have another thing to try tell me.
Thanks

alexanderhiam, 2015-12-11:
Hmm, it looks like pip didn't actually uninstall it... try manually deleting it:

    # rm -rf /usr/local/lib/python2.7/dist-packages/PyBBIO*
    # rm -rf /usr/local/lib/python2.7/dist-packages/bbio*

(Being very careful not to `rm -rf` the wrong thing since you're running as root!)

Then make sure it's really uninstalled:

    # cd ~
    # python -c "import bbio; bbio.__file__"

You should see something like:

    Traceback (most recent call last):
    File "<string>", line 1, in <module>
    ImportError: No module named bbio

If not it'll print the path to where PyBBIO is installed, make sure to `rm` that as well (sometimes if you've installed from a git clone and from pip you'll end up with it in multiple locations).

Then try the pip install again. The install process should create the directory `/root/.BBIOServer/` - if you still don't have it after that then there could possibly be something wrong with the install script, though I just tested a fresh install and it created it for me.

Rocketpol, 2015-12-14:
Thanks alexander now it work, the only thing that now doesn't work is the toggle of led USR2 that show always current state: 255. while USR3 LED works fine.