graycatlabs/PyBBIO #93: importing bbio causes immediate segfault

jasedit, 2015-12-18:
I've got a Beaglebone Black, freshly installed with Ubuntu 14.04.3 (kernel 4.1.13-ti-r36), and I've downloaded this repository and run the doinstall.sh script to install the library. Whenever I attempt to run any of the examples, they do nothing, and adding in print statements to check operation produces no output. If I load up an interactive python session, and attempt to use the library, I get the following behavior:

```
root@luggage:~# python
Python 2.7.6 (default, Jun 22 2015, 18:31:26) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> import bbio
Segmentation fault
root@luggage:~# python
Python 2.7.6 (default, Jun 22 2015, 18:31:26) 
[GCC 4.8.2] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from bbio import *
Segmentation fault
```

I've tried running it as a normal user as well, to no avail. Is there a step I'm missing to get the library up and running?

alexanderhiam, 2015-12-18:
That's because PyBBIO doesn't support 4.x kernels yet, but it's in the works.

You could try downgrading to 3.8, though I'm not sure how to do that on the Ubuntu image (I mainly just target the official BeagleBoard.org Debian images).

If you're not attached to Ubuntu and 4.x you could also load the Debian 7.9 image, which has the 3.8 kernel, from here: http://beagleboard.org/latest-images

jasedit, 2015-12-18:
I need Ubuntu in order to be able to interface with ROS - I was hoping to use a more modern kernel, since we've had a lot of trouble with older kernels (I was in the process of attempting to get a working system with a newer kernel.) I know how to downgrade the kernel, so I can give that a shot as well.

alexanderhiam, 2015-12-18:
OK, that would be good to confirm things are still working on Ubuntu with 3.8 (you shouldn't have any trouble with that...). 

The best case ETA for 4.x support would be within the next week as I'll have some time to work on it, though that's assuming no big, unforeseen issues emerge.

Mausy5043, 2016-04-17:
I'm running into the same issue. Any progress?

patrmccormack, 2016-04-25:
Same problem on Debian 8.3 -- is this still being actively developed?

alexanderhiam, 2016-04-26:
@patrmccormack I'm working on it when I can - you can track the progress (or help out) here: https://github.com/graycatlabs/PyBBIO/milestones/4.1%20kernel%20support

patrmccormack, 2016-04-26:
Thanks Alex
I'll be delivering this project in two weeks time. From 27th May I'll have
some time to spare and would be happy to help.
Pat

“A goal without a plan is just a wish.”
― Antoine de Saint-Exupéry
<http://www.goodreads.com/author/show/1020792.Antoine_de_Saint_Exup_ry>

jlewallen18, 2017-09-26:
Any chance this can still be completed? I'm interested in this, and when I run the bbio import I also get an immediate segfault