graycatlabs/PyBBIO #95: GPIO does not work after "bbio.pinMode"

heylouiz, 2016-01-05:
I'm having a problem initializing a GPIO in Beagle Bone Black.

If I do this in the Python interpreter:
```
 bbio.pinMode(GPIO_PIN, bbio.OUTPUT)
 bbio.digitalWrite(GPIO_PIN, bbio.HIGH)
```
The GPIO does not respond to value variations.

But if i do this:
```
 bbio.pinMode(GPIO_PIN, bbio.OUTPUT)
 bbio.pinMode(GPIO_PIN, bbio.INPUT)
 bbio.pinMode(GPIO_PIN, bbio.OUTPUT)
 bbio.digitalWrite(GPIO_PIN, bbio.HIGH)
```
The GPIO starts to work normally.

This problem didn't occur in a older white Beagle Bone.

Any thoughts?

Thanks!

GingerMuda, 2016-02-27:
It seems the guys who developed the library are not supporting it. Can you try the Adafruit BBIO, they can get your attention real quick.

alexanderhiam, 2016-02-27:
@GingerMuda there's only one of me, and I'm doing my best to support it. The library is open source, so any contributions are welcome, feel free to pick away at the [issues](https://github.com/graycatlabs/PyBBIO/issues) if you'd like to help out...

@heylouiz which GPIO pin(s) is this happening with? Do you have the latest PyBBIO version installed? i.e.:

    # pip install -U PyBBIO

heylouiz, 2016-03-01:
@alexanderhiam I remember trying to update the software, so I believe it was the last pip version, I didn't tried to use the github version though..
I didn't investigate any further, cause the problem was solved using the workaround I described in the issue.
The GPIOs are GPIO2_22 and GPIO2_23.