graycatlabs/PyBBIO #97: no module named WebCam

GingerMuda, 2016-02-27:
Hi,

Tried running the webcam_bbioserver_test.py but getting an import error "no module named WebCam". GStreamer is installed as instructed.
Both webcam_test and webcam_bbioserver_test.py files are in the same folder.
What am i missing out?

alexanderhiam, 2016-02-27:
Are you running the example from in `/usr/local/lib/PyBBIO/examples/`?  If so make sure you have the latest version of PyBBIO:

    # pip install -U PyBBIO

then try again.

If you're running from insides a clone of the repo, have you done a `# git pull` to make sure you have the latest code?

GingerMuda, 2016-02-27:
I'm a newbie, i copied the whole code into a file using nano, created  .py files and placed them in the same folder but when running them, i still get the same error. I also tried the pip command but highlighted that i have the latest library. I cannot access the examples using that file path, i'm getting a bash error.

alexanderhiam, 2016-02-27:
What image are you running on the BeagleBone? What's the output of `# uname -a`?

If you post the entire output you get when you try to run the script I'll have a better idea of what's going on.

GingerMuda, 2016-02-27:
The output is:  Linux beaglebone 3.8.13-bone79 #1 SMP Tue Oct 13 20:44:55 UTC 2015 armv7l GNU/Linux

Some info that might be relevant: I am accessing the BBB through ssh, i don't have a desktop view. I also have an internet connection through the same USB cable.

root@beaglebone:~# uname -a
Linux beaglebone 3.8.13-bone79 #1 SMP Tue Oct 13 20:44:55 UTC 2015 armv7l GNU/Linux
root@beaglebone:~# cd Project
root@beaglebone:~/Project# python webcam_bbioserver_test.py
Traceback (most recent call last):
  File "webcam_bbioserver_test.py", line 8, in <module>
    from WebCam import WebCam
ImportError: No module named WebCam
root@beaglebone:~/Project# 

When i also try to import it in the Python Interpreter (see below);

root@beaglebone:~/Project# python
Python 2.7.3 (default, Mar 14 2014, 17:55:54) 
[GCC 4.6.3] on linux2
Type "help", "copyright", "credits" or "license" for more information.
>>> from bbio import *
PyBBIO initialized
>>> import WebCam
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: No module named WebCam
>>>

alexanderhiam, 2016-02-27:
Ah, ok, you have some older code from before the libraries got moved inside the `bbio` package itself. The import line should be:

    from bbio.libraries.WebCam import WebCam

See the current version here: https://github.com/graycatlabs/PyBBIO/blob/master/examples/webcam_bbioserver_test.py

GingerMuda, 2016-02-27:
This is the error i am getting after running the server   .py file;

root@beaglebone:~/Project# python webcam_bbioserver.py
Traceback (most recent call last):
  File "webcam_bbioserver.py", line 10, in <module>
    from bbio.libraries.BBIOServer import *
  File "/usr/local/lib/python2.7/dist-packages/bbio/libraries/BBIOServer/__init__.py", line 3, in <module>
    from bbio_server import *
  File "/usr/local/lib/python2.7/dist-packages/bbio/libraries/BBIOServer/bbio_server.py", line 26, in <module>
    os.chdir(ROOT_DIR)
OSError: [Errno 2] No such file or directory: '/root/.BBIOServer'
root@beaglebone:~/Project#

alexanderhiam, 2016-02-27:
Hmm, `/root/.BBIOServer` should get created during the installation. Try uninstalling and reinstalling it:

    # pip uninstall PyBBIO
    # pip install PyBBIO

GingerMuda, 2016-02-27:
Thank you very much its now working. It has a latency of about 10-12s, is there any way i can reduce that besides changing the resolution of the camera? Because right now it is running at 1280*960. 
When i uninstalled it, the file path still existed. I tried removing it as root/ but it failed so i had to change directory to /.

alexanderhiam, 2016-02-27:
Glad to hear it.

I haven't found a way around the latency yet, it seems to be inherent in the way it's being done now.

closed 2016-02-27 by alexanderhiam