graycatlabs/PyBBIO #98: Problem running PyBBIO ADC code on BBB attached to 4D Systems 4DCAPE

akerbay, 2016-03-15:
Alex:
I just now installed your PyBBIO and am having a problem with analogRead. My BBB is attached to a 4D_Systems 4DCAPE, and apparently that's causing the problem. Have a similar problem with Adafruit ADC code. Is the problem fixable? (BTW, PyBBIO PWM output works fine.)

http://www.4dsystems.com.au/product/4DCAPE_70T/

Traceback (most recent call last):
  File "/var/lib/cloud9/ADC_Test.py", line 23, in <module>
    run(setup, loop)
  File "/usr/local/lib/python2.7/dist-packages/bbio/bbio.py", line 60, in run
    loop()
  File "/var/lib/cloud9/ADC_Test.py", line 13, in loop
    val1 = analogRead(pot1)
  File "/usr/local/lib/python2.7/dist-packages/bbio/platform/beaglebone/adc.py", line 30, in analogRead
    cape_manager.load(ADC_ENABLE_DTS_OVERLAY, auto_unload=False)
  File "/usr/local/lib/python2.7/dist-packages/bbio/platform/beaglebone/cape_manager.py", line 27, in load
    f.write(overlay)
IOError: [Errno 17] File exists

Here's my program:
# Import PyBBIO library:
from bbio import *

pot1 = AIN4 # pin 33 on header P9
pot2 = AIN5 # pin 36 on header P9

def setup():
  # Nothing to do here
  pass

def loop():
  # Get the ADC values:
  val1 = analogRead(pot1)
  val2 = analogRead(pot2)
  # And convert to voltages:
  voltage1 = inVolts(val1)
  voltage2 = inVolts(val2)
  print " pot1 ADC value: %i - voltage: %fv" % (val1, voltage1)
  print " pot2 ADC value: %i - voltage: %fv\n" % (val2, voltage2)
  delay(500)

# Start the loop:
run(setup, loop)

Thanks for your help.
John

alexanderhiam, 2016-03-16:
I haven't used the 4D systems cape, but it looks like it's got a resistive touch screen and is using the ADC inputs AIN1, AIN2, AIN3 and AIN4 for that. 

In theory it would be possible to get AIN5 and AIN6 working for general use, but that would require some Device Tree hacking, and maybe editing the DT overlay used by that cape. If you need the cape and some ADC inputs, it would probably be a lot easier to just use an external ADC. e.g. PyBBIO already has a library for the [ADS786x](https://github.com/graycatlabs/PyBBIO/wiki/ADS786x) series SPI ADCs.

akerbay, 2016-03-16:
Thanks Alex. Late last night after updating the Debian OS to from 3.8.13
bone47 to bone79, the PyBBIO ADC and PWM routines work 😊 but now the LCD
screen isn't showing the usual Debian stuff, only an arrow cursor that I
can move around. And also the Linux penguin shows briefly. So the LCD is ok
and I suppose there's some conflict between the PyBBIO software and the
screen software in the Debian distro. I'm hoping to get some answer from 4D
Systems about it or maybe from Robert Nelson of Debian.  So there may be
some hope to getting it all working. I'm getting pretty desperate because I
have to deliver some control systems soon and need both the PyBBIO and the
LCD working.
Thanks very much,
John

alexanderhiam, 2016-03-16:
Well that sounds like progress.

Unfortunately I doubt you'll get much support from 4D, considering the datasheet is wa outdated with only Angstrom instructions, and has this notice:

> 4D Systems does NOT support any software associated with the BBB nor software and drivers associated with this CAPE. It is up to the user to determine which software is used on the BBB and which distributions are used on the BBB, and therefore which are compatible with this CAPE.

:unamused: 

You might be able to get some help on the #beagle irc channel (Robert is rcn-ee in there).

akerbay, 2016-03-16:
Yes, I'm pretty sure you're right about 4D and I do think Robert is a 
good possibility for help. I emailed him but haven't heard back yet. And 
I do like your idea about thanks again.the SPI ADC and DAC. I was hoping 
not to have to do that, wanting to take advantage of the BBB's ADC and PWM.
Thanks again,
John
http://johnbakeree.blogspot.com/

akerbay, 2016-03-17:
Hi Alex:
I reverted to bone50 (Debian 3.8.13-bone50), reinstalled the PyBBIO 
software and apparently both the PWM and ADC routines work. Perhaps I 
had simply improperly installed bone79 or some other newbie problem. :-[
Thanks,
John

alexanderhiam, 2016-03-18:
Weird, so the ADC is working along side the LCD and touchscreen? 

It could be that the overlay the 4D cape is using changed after bone50. Since that overlay was made for different hardware and 4D offers zero software support, I wouldn't be surprised if it's just totally broken in more recent kernel versions :unamused:

akerbay, 2016-03-18:
Hi Alex,
I may have misspoken. With the Debian bone50, the LCD and both the software for the ADC and PWM are working fine. That all seems to be in good shape but unfortunately I can't run my Python program because one of the Tkinter routines isn't getting installed, TkAgg. TkAgg is required for my GUI and it had previously installed on the previous Debian, bone47, and except for the ADC and PWM everything had run so I should be able to install it on this Debian. Maybe a newbie goof. I will try bone79 again. I have been searching for a solution to the missing TkAgg. Others had reported the same problem but I haven't found the answer. 
Regarding 4D, they don't provide any  software but strangely were recommending Android in their data sheet. I don't think Android has ever been used on the BBB, just originally Angstrom and now Debian and Ubuntu. 
Anyhow it's all on me to get it running. 
Fortunately your PyBBIO works great and I should be able to get it all operational. 
Thanks for your help,
John

akerbay, 2016-03-26:
Alex:
I found a posting on the Beagleboard Forum that suggested connecting a 
keyboard and mouse to the BBB in order to get a program running. 
Surprisingly this worked and my Python matplotlib tkinter PyBBIO GUI 
program is now working. Just a few more details. Don't know why the 
keyboard and mouse are required but I'm planning on making my program 
startup on reboot, so won't need the keyboard and mouse in the final 
systems. That's a detail to be worked out.
Thanks for your help,
John