graycatlabs/PyBBIO #99: not an issue but a doubt

iamharshit, 2016-03-16:
just wanted to know like we can use external interrupt ,can we use timer interrupt too in beagle?

alexanderhiam, 2016-03-16:
Well, that depends what kind of accuracy and resolution you need. There's plenty of ways to do timed tasks in GNU/Linux, like cron, or event-driver frameworks (e.g. Twisted or Tornado for Python).

iamharshit, 2016-03-16:
actually i was working on implementing a rpm-encoder on beagle using PyBBIO, but for that i needed external interrupt and timer interrupt , although i got to know how to use external interrupt in PyBBIO, but still stuck on how to use timer interrupt in PyBBIO. any direction from you?

alexanderhiam, 2016-03-16:
Have you looked at using the eQEP modules? They can very precisely count steps from an encoder.