graycatlabs/PyBBIO #101: Unable to use PWM1A and PWM1B at the same time

raunaqbhirangi, 2016-06-19:
I'm trying to use both PWM1A and PWM1B. I am not changing the frequency.
I just use 

analogWrite(PWM1A,200,RES_8BIT)
analogWrite(PWM1B,200,RES_8BIT)

I am able to digitalWrite to both pins when I use them as GPIO1_18 and GPIO1_19. But the analogWrite doesn't work for PWM1B

alexanderhiam, 2016-06-20:
Right, at some point I think the PWM driver changed a little bit, so the way PyBBIO currently does PWM became a little bit wrong :unamused: 

I'm planning to completely rewrite the PWM code as a C extension (issue #62), and I've been reluctant (and busy) to try and debug it before then...

I *think* you can get around it by first explicitly setting both channels to a 0% duty cycle before using them, i.e.:

```python
analogWrite(PWM1A, 0)
analogWrite(PWM1B, 0)

analogWrite(PWM1A,200,RES_8BIT)
analogWrite(PWM1B,200,RES_8BIT)
```

Give that a try and let me know what happens...

MatthewWest, 2016-08-10:
@alexanderhiam I just wrote an update for Adafruit's BBIO library in C, parts of which may be instructive. The relevant file is [this one](https://github.com/MatthewWest/adafruit-beaglebone-io-python/blob/master/source/c_pwm.c)