graycatlabs/PyBBIO #102: Analog In not recognized

Crawforz, 2016-07-03:
For some reason, whenever i try to use any of the analog in pins, it says it does not exist. The digital pins work just fine but cannot figure out why it's doing this, any help would be greatly appreciated.

alexanderhiam, 2016-07-04:
That's odd... does it happen with the `analog_test.py` example program? If it's just your own code that's not working then post that here.

What image are you using? And what's the output of `$ uname -a`?

Crawforz, 2016-07-04:
Yeah, it doesn't work in any program that references an analog pin; it just gives "Value Error: invalid GPIO pin". The disk image is the Wheezy debian image. The output for uname- a gives "Linux beaglebone 3.8.13-bone80".

thanks for the help!

Crawforz, 2016-07-04:
oh wait! my bad, so i copied the program but didn't pay much attention to the set up and assumed that you had to identify the pin as an input (not sure why) and that must have messed it up cause it works now when i just do the "pass" keyword in the setup.  Thanks! This actually still helped me to look at it again, sorry !

closed 2016-07-04 by Crawforz

alexanderhiam, 2016-07-07:
Glad you got it working!